apballard/Marlin #1: Different screen sizes

MarPiRK, 2017-05-20:
Hi, I really appreciate your project and work on integrating Nextion screen with Marlin.
The problem is that I have NX4024T032 model which is smaller screen size than one used in this project.

I'd like to ask if you have any kind of source vector images, so I could work on port to my screen size. I can always downscale JPGs from current hmi project, but I think that the most elegant way of doing it would be to have some SVG folder, so it would be easier to port it to other Nextion screens.

apballard, 2017-09-29:
@MarPiRK Sorry, only saw your request now.

I will upload the source images and what-not as soon as possible, although you most probably no longer need it, i will make it available.

MarPiRK, 2017-09-29:
Sure I'm still interested in it. It's sort of side project, so it doesn't really matter.

weed2all, 2017-10-25:
Hello!anyone still working on nextion display firmware...if you are so kind i can have the firmware for my 3.5"nextion LCD that I want to apply to my hypercube core xy 3d printer using marlin and ramps 1.4