apballard/Marlin #4: Not working on latest Marlin

sputnicker, 2018-05-03:
When I compile I get an error on the serial communication:
I already changed the serial3 to serial2 on nexconfig.h. Nextion board is connected to serial 2 on ramps (pin 16 and 17).

Is this fork still working?
Thank you

```
sketch\Experimental_LCD.cpp: In function 'void processBuffer(const char*)':

Experimental_LCD.cpp:92: error: 'SERIAL_ECHO_START' was not declared in this scope

   SERIAL_ECHO_START;

   ^

Experimental_LCD.cpp:94: error: 'SERIAL_EOL' was not declared in this scope

   SERIAL_EOL;

   ^

sketch\Experimental_LCD.cpp: In function 'void ProcessPage(char*, uint8_t)':

Experimental_LCD.cpp:229: error: 'SERIAL_ECHO_START' was not declared in this scope

  SERIAL_ECHO_START;

  ^

Experimental_LCD.cpp:231: error: 'SERIAL_EOL' was not declared in this scope

  SERIAL_EOL;

  ^

sketch\Experimental_LCD.cpp: In function 'void processMessage(const char*)':

Experimental_LCD.cpp:248: error: 'SERIAL_ECHO_START' was not declared in this scope

  SERIAL_ECHO_START;

  ^

Experimental_LCD.cpp:250: error: 'SERIAL_EOL' was not declared in this scope

  SERIAL_EOL;

  ^

exit status 1
'SERIAL_ECHO_START' was not declared in this scope
```

weeelfi, 2018-06-01:
Just put open and close parenthesis on all of those lines

Mechanically, 2018-08-08:
Thanks weeelfi.
Now I'm having an issue with "lcd_refresh(); was not declared in this scope"
This function doesn't exist elsewhere in the code. Should I just comment it out?

GuusvanMarle, 2018-10-20:
Hello,
I've had the same issue with "lcd_refresh(); was not declared in this scope", after fixing loads of errors before that, and indeed it does not talk to the display. 
Any progress on this error yet?

GuusvanMarle, 2018-10-20:
Lcd_refresh() is also called in UBL_G29.cpp
But i haven't figured out the difference between the two.

neocogent, 2019-03-12:
Any progress on this issue? I may want to update this to work with recent marlin for my own use. If solved, did you create a fork with fixes that may allow building on going forward?