apballard/Marlin #7: Req. A bit of detail on what you did and how to use.

neocogent, 2019-03-12:
I'm thinking of replacing my TFT32 on 3D printer (using Marlin) and adding a new one to a CNC machine (also Marlin based). I haven't bought a display yet but may do if I can figure out how much work is involved to use it.

Your project looks very interesting and may be useful. I can't quite make sense of what you've done. Did you just create a GUI for Nextion or did you also have to add code to Marlin for protocol support? Would this be hard for me to update to a more recent Marlin or just a bit or merge housekeeping? I have strong firmware programming skills but don't necessarily want to spend weeks on getting it to work for me. 

How much time did it take from scratch to do the Nextion GUI work? I may have to edit that to make a CNC suitable display layout. How happy are you with the Nextion overall? I hope you may update the readme with some comments along this line that may make it easier to build on this project. But I know this isn't a recent project or your day job. Just seeing what I can find out that helps. Thx.

reddomon, 2019-04-18:
I'm also thinking of making a new control panel for marlin 2 with Nextion display. Bu i cannot find any clue which c and h file have to change. I made some progress with the g code parser but that is all. I am totally lost. By the way @neocogent i have some design files for nextion 5 inch display. You can find it here [https://github.com/reddomon/vektorial](url)

reddomon, 2019-09-19:
Hi, in case anybody want to use,I used ui_api.h that has lots of functions and structures for ui develeopment. I am develeoping nextion touch controller for Marlin 2.x and it will be ready soon.
[image]

UnknownHero99, 2019-09-25:
> 
> 
> Hi, in case anybody want to use,I used ui_api.h that has lots of functions and structures for ui develeopment. I am develeoping nextion touch controller for Marlin 2.x and it will be ready soon.
> [image]

hmm looks really nice any way to get source code ?

reddomon, 2019-09-26:
Yes, It will be opensourced but not ready yet. I will share the details when it is ready to be published.

reddomon, 2020-02-23:
Here is the first release. [https://github.com/reddomon/Marlin-Nextion-Ui](url)

elifhalegokcen, 2021-03-14:
The link isn't working. Could you share it again